acamilo/RobotInterfaceBoard #21: Fuse for the 5v regulator is needed (Value is in question)

madhephaestus, 2018-03-28:
A fuse to gracefully fail after the 5v rail is plugged directly into gnd, or if the modules 3.3v rail is shorted directly to ground. I assume 75% of the regulators maximum output is sufficient. Be sure to also look at the power spec for the ESP development boards built in. This regulator has fried on a large-ish number of our testing boards. Changing those fusing is much better than replacing all of the power regulators *when* the students short power to ground.

madhephaestus, 2018-03-28:
An additional LED for the 3.3v rail should be here added. I will need to be able to power the module on and see at a glance if all the power rails are up and running.

acamilo, 2018-04-13:
commit 5c16eed2d937238d69e448bf987ce9fe361f2ba1

closed 2018-04-13 by acamilo